hamuchiwa/AutoRCCar #189: How to achieve low latency for video streaming for rc_driver.py?

MyFirstRoboticProject, 2019-04-04:
I'm using mobile hotspot for connecting my raspberry pi and windows PC. Getting a 3 second delay. Anyway to solve it?

MyFirstRoboticProject, 2019-04-05:
@hamachiwa 
Did you use stream.py or stream_fast.py for streaming in raspberry pi? I'm using stream_fast.py, so would using the stream.py fix the issue? Any advice would be appreciated

MyFirstRoboticProject, 2019-04-05:
@DevanShice020 Is your streaming working fine rc_driver.py. I'm having a 3 sec delay

hamuchiwa, 2019-04-05:
@MyFirstRoboticProject `stream.py` was used, the fast one was added later as an alternative. Both of them are taken from picamera documentation. 
Have you tried using a router instead? Maybe it's the hotspot causes the latency. . 
In case you haven't seen this [video](https://www.youtube.com/watch?v=KrvvmLEQk8o) yet, try to slow down your car as well. 
In the worst case if the delay cannot be improved, perhaps you can let the car move only after receiving an image, then car stops and waits for the next frame, then moves again.

Devanshice020, 2019-04-05:
But sir how can we control stopping the image from receiving the pi camera
i mean it will continuous capture image to run on the track (rc._drive.py)
so how is it possible to stop the car , receive an image, and then allow
the car to run again @ hamuchiwa

hamuchiwa, 2019-04-05:
@Devanshice020 We are not stopping streaming, we could stop the car after each move and let it wait until next frame. Adding `self.rc_car.stop()` after `self.rc_car.steer(prediction)`. Give it a try and hope it could easy the high delay issue.

MyFirstRoboticProject, 2019-04-05:
@hamuchiwa
Well I'll give that a try. Thanks! Also my car needs three AA batteries for power so would replacing one of the AA batteries with a simple copper wire do the trick? It would reduce the effective voltage from 4.5 V to 3 V so the car should slow down. Will it work or would it harm the car?

Devanshice020, 2019-04-05:
@hamuchiwa it worked.
Is it possible : i have more powered car rock crawler 1:18 and i did the same as you mentioned but my car did not moved an inch .. so what do you think might be the problem? Does the power of the car has something to do with the car not moving forward?

hamuchiwa, 2019-04-05:
@Devanshice020 @MyFirstRoboticProject Be experimental may help...? For example, add some delay between steer and stop, if the car moves then it's the newly added line, if not, it's the car itself.

Devanshice020, 2019-04-05:
how to give delay? @hamuchiwa 
Thanks in advance Sir

VishalWrench, 2019-04-06:
import time
time.sleep(time in sec)

closed 2019-04-06 by MyFirstRoboticProject

MyFirstRoboticProject, 2019-04-06:
@Devanshice020
Did the battery method work?

Devanshice020, 2019-04-06:
Actually my car has rechargeable lithium battery so this method will not work for that

Devanshice020, 2019-04-06:
You can try this it worked for me but after that car speed will be very low but my objective for detecting lane for forward, left and right
https://github.com/hamuchiwa/AutoRCCar/issues/189#issuecomment-480295132