hamuchiwa/AutoRCCar #194: Track following issue

TejasSaralaya, 2019-04-10:
If the car is kept in any other angle other than straight then does it follows the track???

vishwasnayak597, 2019-04-12:
Even i had the same issue. Trust me dude this codes will not work. You have to modify it. Its so difficult to implement, so i suggest you to change the project itself.  You will never get solution even if you try for an year.This code doesnt even detect lane ! so whats point of doing it? Just think about it.

vishwasnayak597, 2019-04-12:
@TejasSaralaya Even i had the same issue. Trust me dude this codes will not work. You have to modify it. Its so difficult to implement, so i suggest you to change the project itself. You will never get solution even if you try for an year.This code doesnt even detect lane ! so whats point of doing it? Just think about it.

krukhlis, 2019-04-12:
Guys, this project is not enterprise-graded AutoPilot for cars. It's a PoC, for specific use cases with clearly defined conditions. It does exactly what it states. Nothing else. If you want to add new functionality -- you are always welcome, it's open sourced, PRs are allowed. Or do you really think that someone will waste his life developing fully functional AP  and will opensource it and give it to you for free? C'mon, be realistic.

vishwasnayak597, 2019-04-12:
I am not sure about that

akbaraliseo, 2024-04-29:
May you need to read this blog https://whatinmind.com/netwyman-blogs/ because its difficult to replay directly