hamuchiwa/AutoRCCar #196: Can you explain this part of code for traffic light detection

Devanshice020, 2019-04-18:


hamuchiwa, 2019-04-18:
@Devanshice020 
This "for loop" is used to draw bounding boxes for the detected object. The height, width along with x, y positions are returned by cascade_obj, with those you can draw a bounding box. 
Because we used two cascade classifiers (stop sign and traffic light), and we also know the bounding box of stop sign is a square, traffic light is a rectangle (based on training data shape), so we can simply separate these two by calculating `width/height`, then put different texts on top of the box.

Devanshice020, 2019-04-18:
Then for traffic light I have to measure size of my traffic light box?

hamuchiwa, 2019-04-18:
@Devanshice020 No need to measure, because there are only two objects, if `width / height == 1` then it's a square, if not it's a rectangle.

Devanshice020, 2019-04-18:
I trained my own classifier for traffic light but when camera detects red or green light some times it detects and some times it does not detect. What changes should I do?

Should I change this part of code ?

# check if light is on
            if maxVal - minVal > threshold:
                cv2.circle(roi, maxLoc, 5, (255, 0, 0), 2)

                # Red light
                if 1.0 / 8 * (height - 30) < maxLoc[1] < 4.0 / 8 * (height - 30):
                    cv2.putText(image, 'Red', (x_pos + 5, y_pos - 5), cv2.FONT_HERSHEY_SIMPLEX, 0.5, (0, 0, 255), 2)
                    self.red_light = True

                # Green light
                elif 5.5 / 8 * (height - 30) < maxLoc[1] < height - 30:
                    cv2.putText(image, 'Green', (x_pos + 5, y_pos - 10), cv2.FONT_HERSHEY_SIMPLEX, 0.5, (0, 255, 0),2)
                    self.green_light = True

                # yellow light
                # elif 4.0/8*(height-30) < maxLoc[1] < 5.5/8*(height-30):
                #    cv2.putText(image, 'Yellow', (x_pos+5, y_pos - 10), cv2.FONT_HERSHEY_SIMPLEX, 0.5, (0, 255, 255), 2)
                #    self.yellow_light = True

hamuchiwa, 2019-04-18:
@Devanshice020 Not really, the code is just for adding text, if the classifier doesn't detect the object in the first place, there won't be a bounding box to draw.

Devanshice020, 2019-04-19:
Should I retrain my traffic_light_claassifier classifier ?
Is there any alternate option for classifying traffic light in windows?
[image]
This is a traffic light I have made plz help
Thanks in Advance

hamuchiwa, 2019-04-19:
@Devanshice020 The traffic light classifier was trained only with images of that small toy you see in the video. I think you may give it a try to make the shape of your traffic light look like that toy, before training your own.

closed 2019-04-25 by Devanshice020