hamuchiwa/AutoRCCar #199: Has anyone finished this project?

zhushengtong, 2019-04-22:
Has anyone finished this project?I find a lot of people have problems

hamuchiwa, 2019-04-23:
@zhushengtong 这个项目好多年了，提问的人也不少，所以大概率是有的。我能找到的视频链接有三个，你搜一下issue里面也能找着。估计一般情况是做完就完了，要么是试了一下效果不好就放下了。
最后感谢提供视频的小哥们。
https://www.youtube.com/watch?v=KrvvmLEQk8o
https://www.youtube.com/watch?v=PFKm-EuyuWo
https://www.youtube.com/watch?v=AVuk-TeDnvE

zhushengtong, 2019-04-25:
谢谢你的回答~
  可能是我的知识储备不太够，所以请您见谅，我是一个本科生。我想了一阵子，也不太知道您的数据集是怎么采集出来的。并且我看了几个之前的issue。
  如果我想完成前进这个操作，我是否应该不止采集车头完全朝向前方的图片，因为在运行过程中，可能出现跑偏，这时候需要进行校正，所以这时候我是否还需要多采集车头跑偏的照片，然后进行训练。并且在弯道中肯定也会涉及到需要直走的情况。
  并且我想知道，为什么大佬没有用传统的特征点提取，并直线拟合的方法呢？
  这个问题困惑了我一阵子了。看到您的这个项目是三个月前，之前对神经网络还不了解，但是现在即使很了解了之后，在真正的实操过程中还是不知道该怎么去完成，期望能得到您宝贵的帮助，谢谢~
我十分希望能得到您当初训练时用到的数据集，这样我才能知道您是怎么采集图像的。
  如果您能帮助我，我不胜感激~~
  如果您愿意共享一下您的数据集的话，这是我的邮箱zhushengtong@163.com
  谢谢！

hamuchiwa, 2019-04-25:
@zhushengtong 
1. 你说的都对的，其实收集数据的话来回多跑几遍就好，边边角角的情况也就基本上涵盖进去了。还是不行的话，再收集一些特殊情况的数据。
2. 谈不上什么大佬，大家都是边学习边实践嘛。当时做这个项目就是为了用一下神经网络，相当于走了个捷径，PID控制也省掉了，神经网络直接出结果。但如果真实际应用起来，我觉得还是车道检测来的实在。Nvidia之前有篇[文章](https://images.nvidia.com/content/tegra/automotive/images/2016/solutions/pdf/end-to-end-dl-using-px.pdf)，是用图像来预测方向盘转弯角度，有兴趣的话你可以看一下。
3. 图片的话，你可以看一下`test/model_train_test/`里的那个jupyter notebook，实际的收集和训练只取的下半部分。

最后贴两个链接，这两个项目都挺棒的，第二个还包括车道检测：
https://github.com/pseudoyim/galvaneye
https://github.com/jawilk/Self-Driving-RC-Car-Payment

zhushengtong, 2019-04-26:
谢谢你每一次认真的回答我的疑惑~我觉得第二个项目很棒，正在钻研~