hamuchiwa/AutoRCCar #203: rc_driver_nn_only.py runs fine but problem with rc_driver.py

tc2n, 2019-04-27:
My `rc_driver_nn_only.py` file runs fine, but when I try to run `rc_driver.py` and run `stream_server.py` on the client I get broken pipe error on the client side and server command just terminates without giving any error. This error is generated only when I start `stream_server.py` on the client side.

enigma0503, 2019-04-27:
I am also getting the same error(broken pipe error when I run the `stream_server.py`)

codeskul, 2019-04-28:
I face same issue

codeskul, 2019-04-28:
Only ultrasonic_client.py runs when I start stream_client.py it show me error

codeskul, 2019-04-28:
[image]

codeskul, 2019-04-28:
[image]

hamuchiwa, 2019-04-28:
@tc2n @codeskul See if this is what you encountered on the server side: #176, #185

codeskul, 2019-04-28:
@hamuchiwa no sir it not work

hamuchiwa, 2019-04-28:
@codeskul What's the error on the server side?

codeskul, 2019-04-28:
When I start server side code rc_driver.py it don't show any error, in raspberry pi i start ultrasonic_client.py it sends sensor data well, as I start stream_client.py on raspberry pi it stop and show broken pip error on raspberry pi as shown above and at server side it shows system exit as shown above

codeskul, 2019-04-28:
It stop server side execution when i start raspberry side stream_client.py

[image]

hamuchiwa, 2019-04-28:
@codeskul Have you tried running stream_client first or on linux? This project was only tested on Linux and Mac, it could the way of handling threads is different on windows.

codeskul, 2019-04-28:
@hamuchiwa  ok i will try it on Linux and reply to you

tc2n, 2019-04-29:
@hamuchiwa I face the same problem on Linux.

tc2n, 2019-04-29:
I tried to remove the 'stop sign' detection part from the `rc_driver.py` file, then I was able to run `stream_server.py` file too, but still when I try to run `ultrasonic_stream` file I got Broken Pipe error with no error on Server Side (I am unable to run both `stream_server.py` and `ultrasonic_stream.py` simultaneously).

hamuchiwa, 2019-05-02:
@tc2n @codeskul My mistake, forgot to put `self` before variables in last update. `rc_driver.py` has been updated and should be working now.

closed 2019-05-02 by hamuchiwa

ShwetaLalwani, 2019-05-09:
@tc2n Hi, is your car moving in the correct directions after running rc_driver_nn_only.py?

tc2n, 2019-05-10:
@ShwetaLalwani When I run model_training on .npz file, sometimes it works fine. But sometimes when I again train the model on the same .npz file my car always moves in the same direction(Always right or Always forward).