hamuchiwa/AutoRCCar #206: Urgent: Port issue

akilystic, 2019-05-06:


akilystic, 2019-05-06:
I get the following error when i run rc_control_test.py 

(auto-rccar) akash@asp:~/Downloads/AutoRCCar-master/test$ python rc_control_test.py
pygame 1.9.4
Hello from the pygame community. https://www.pygame.org/contribute.html
Traceback (most recent call last):
  File "/home/akash/miniconda3/envs/auto-rccar/lib/python3.7/site-packages/serial/serialposix.py", line 265, in open
    self.fd = os.open(self.portstr, os.O_RDWR | os.O_NOCTTY | os.O_NONBLOCK)
PermissionError: [Errno 13] Permission denied: '/dev/ttyACM0'

During handling of the above exception, another exception occurred:

Traceback (most recent call last):
  File "rc_control_test.py", line 72, in <module>
    RCTest()
  File "rc_control_test.py", line 14, in __init__
    self.ser = serial.Serial("/dev/ttyACM0", 115200, timeout=1)           # linux
  File "/home/akash/miniconda3/envs/auto-rccar/lib/python3.7/site-packages/serial/serialutil.py", line 240, in __init__
    self.open()
  File "/home/akash/miniconda3/envs/auto-rccar/lib/python3.7/site-packages/serial/serialposix.py", line 268, in open
    raise SerialException(msg.errno, "could not open port {}: {}".format(self._port, msg))
serial.serialutil.SerialException: [Errno 13] could not open port /dev/ttyACM0: [Errno 13] Permission denied: '/dev/ttyACM0'

kiran1115, 2019-05-06:
are you using windows ?

akilystic, 2019-05-06:
No.. I'm using Linux Ubuntu

akilystic, 2019-05-06:
I was having similar port permission issue while burning Arduino code but somehow managed to fix it.
Please help me to fix this issue now. Pygame opens and closes abruptly. 
Port Permission denied it says

nandhukrishnan, 2019-05-06:
> I was having similar port permission issue while burning Arduino code but somehow managed to fix it.
> Please help me to fix this issue now. Pygame opens and closes abruptly.
> Port Permission denied 
[image]

You first check what is currently selected for the port in Arduino IDE. Then give the same instead  of _/dev/ttyACM0_  in rc_control_test.py. Doing this just solved my problem.

akilystic, 2019-05-06:
Now serial port isn't visible in Arduino ide. It shows the same Port in the bottom left corner though. Thanks.. I'll try to fix this and let you know

closed 2019-05-29 by hamuchiwa