hamuchiwa/AutoRCCar #211: 关于模型的训练

BreakerMark, 2019-05-23:
您好，我在模型训练时遇到了一些问题。我收集了前进、左转、右转分别1000帧左右，然后分别训练之后可以得到正确的结果，但是将所有的数据放在一起训练得到的模型，系统提示的准确率在90%以上，但是在实际测试时无法得到正确的反馈。请问这个问题应该怎么解决，谢谢。

hamuchiwa, 2019-05-23:
@BreakerMark 一起训练就好，单独训练不成了三个模型了，训练完了主要看一下Validation accuracy是多少。`model_training.py`只把数据分成了train和valid，其实你也可以更进一步，用sklearn搞一下[grid search](https://scikit-learn.org/stable/modules/generated/sklearn.model_selection.GridSearchCV.html)调调参，看看能不能选出一个比较好的模型。再不行就试一下CNN。

BreakerMark, 2019-05-24:
谢谢回复

BreakerMark, 2019-05-24:
想问一下，我刚刚开始学习这些，这个项目如何能够加入CNN呢。怎么才能把这个项目变成Tensorflow或者其他工具？

hamuchiwa, 2019-05-24:
@BreakerMark 模型训练其实是独立出来的，可以单独创建一个文件使用tensorflow，最后跑车的时候只是加载模型，然后做预测就好。我之前有上传一个使用keras的例子，有兴趣你可以看一下 https://github.com/hamuchiwa/AutoRCCar/issues/173#issuecomment-478619182

BreakerMark, 2019-05-25:
谢谢回复，那如果用keras生成的模型文件，是否需要用keras做预测，还是说也可以导入opencv中直接使用呢？

hamuchiwa, 2019-05-25:
@BreakerMark 对的，要用keras做预测，把和opencv相对应的部分替换一下就可以了。