hamuchiwa/AutoRCCar #217: issue when i run rc_control_test.

harshithdayananda, 2019-06-04:
(auto-rccar) C:\Users\Administrator\Desktop\AutoRCCar-master\test>python rc_control_test.py
pygame 1.9.6
Hello from the pygame community. https://www.pygame.org/contribute.html
Traceback (most recent call last):

  File "rc_control_test.py", line 72, in <module>
    RCTest()

  File "rc_control_test.py", line 13, in __init__
    self.ser = serial.Serial("/dev/tty.usbmodem1421", 115200, timeout=1)    # mac

  File "C:\Users\Administrator\Miniconda3\envs\auto-rccar\lib\site-packages\serial\serialwin32.py", line 31, in __init__
    super(Serial, self).__init__(*args, **kwargs)

  File "C:\Users\Administrator\Miniconda3\envs\auto-rccar\lib\site-packages\serial\serialutil.py", line 240, in __init__
    self.open()

  File "C:\Users\Administrator\Miniconda3\envs\auto-rccar\lib\site-packages\serial\serialwin32.py", line 62, in open
    raise SerialException("could not open port {!r}: {!r}".format(self.portstr, ctypes.WinError()))
serial.serialutil.SerialException: could not open port '/dev/tty.usbmodem1421': 

FileNotFoundError(2, 'The system cannot find the path specified.', None, 3)

harshithdayananda, 2019-06-04:
@hamuchiwa please help my demonstration is due on 13th June.

hamuchiwa, 2019-06-04:
@harshithdayananda The port you see is for Mac, change the serial port to match your Windows one, perhaps 'COM3' in your case.

closed 2019-06-04 by hamuchiwa

harshithdayananda, 2019-06-05:
on which line number is the port for windows? i can only see mac and linux.

hamuchiwa, 2019-06-05:
@harshithdayananda Just replace '/dev/tty.usbmodem1421' with your port 'COM3'. It's normal to have a pygame window open, it's for listening keyboard inputs, you will see some outputs in your console if you run the file in command line.

harshithdayananda, 2019-06-05:
> @harshithdayananda The port you see is for Mac, change the serial port to match your Windows one, perhaps 'COM3' in your case.

@hamuchiwa on which line number is the port for windows? I can only see mac and Linux. also when I change the port for Mac and run the script I get a pi game window with a black screen.
[image]

like this.

Yoganandhan01, 2020-01-29:
For using windows pc i am also having the same error..