hamuchiwa/AutoRCCar #227: Neural network weird behavior!!!

Sirajabudabe, 2019-09-29:
in training even when i don't use reverse button .... i get some predictions of reverse 
training accuracy is to low !! how to reduce NN output to be just (Forward-right-left)???and what is the best attributes for good accuracy ?  
any ideas ?

hamuchiwa, 2019-10-01:
@Sirajabudabe Please see #173 and #187 for help, certainly reverse can be removed (it's not used anyway), you need to modify the label array size wherever you see that in the code.

closed 2019-10-01 by hamuchiwa

Sirajabudabe, 2019-10-02:
@hamuchiwa  thanks .. i just want to ask about the training (collecting data ) should i do the "right-left , forward " in separate npz file  or i collect data over hole track (all directions at once )?
does it effect the training accuracy ?