hamuchiwa/AutoRCCar #239: pygame

HarshRangwala, 2020-01-11:
pygame window always crashes after running a python script for collecting data!
please help!

arunph, 2020-02-06:
Can I know the configuration of your system which you are using to execute it.

HarshRangwala, 2020-02-10:
System config is:
Windows 10
Anaconda, spyder to run this file
On my raspi I am using Thonny IDE to run camera_stream.py

arunph, 2020-02-10:
To get better results you need to use 8GB  RAM system which has atleast of 120GB SSD  and install programs like anaconda and spyder in SSD volume. 

Thanks.

HarshRangwala, 2020-02-10:
Oh I should've mentioned, my bad. I have 16 gigs ram, 250g gigs ssd 1 TB Hard disk. System config seems perfect just wondering if there's a bug or something in the code.

arunph, 2020-02-11:
As per my perspective there is no bug in code.it worked for me perfectly.

closed 2021-04-06 by hamuchiwa