hamuchiwa/AutoRCCar #242: Sending predictions to the Arduino

9salman7, 2020-02-10:
Hi @hamuchiwa , you mentioned that the predictions are the sent to the Arduino for RC car control. How is this done? As far as I understand, the Arduino is only connected to the remote controller. How exactly is the data sent to the Arduino without connecting it to the RPi or the Computer?

Thanks for your help!

arunph, 2020-02-10:
No need to connect raspberry pi to Rc control unit. You can connect Rc control with Arduino. Since raspberry pi will  send the and receive pi camera and ultrasonic sensor data from and to the computer which in further sends Direction instructions to Ardunio to control the Car.

9salman7, 2020-02-10:
How exactly does the computer send instructions to the Arduino? @arunph

arunph, 2020-02-10:
kindly go through this link https://zhengludwig.wordpress.com/projects/self-driving-rc-car/

9salman7, 2020-02-10:
I did go through it, it mentions that the Arduino is connected to the computer via USB. I'm still unsure how the connection is done remotely.

arunph, 2020-02-10:
As the per the project is concern,raspberry pi need to be powered up and then connect its wifi to router or mobile hotspot and then connect your computer wifi  to same network which you have used to connect raspberry pi.Both should lie in the same network.

closed 2020-03-21 by 9salman7