hamuchiwa/AutoRCCar #245: Stop sign detection

9salman7, 2020-03-10:
Hi @hamuchiwa , can you please explain why you have used the following logic for detecting the stop signs?

`            # stop sign
            if width / height == 1:
                cv2.putText(image, 'STOP', (x_pos, y_pos - 10), cv2.FONT_HERSHEY_SIMPLEX, 0.7, (0, 0, 255), 2)`

why does width/height == 1 mean that the object is a stop sign? For reference, those two variables are returned from the detectMultiScale function from OpenCV

maazmmd, 2020-03-10:
Am also stuck here, How to stop the car using a stop sign

hamuchiwa, 2020-03-12:
@9salman7 The bounding box of stop sign in this project is a square, so if width == height, it's a stop sign.

9salman7, 2020-03-13:
> @9salman7 The bounding box of stop sign in this project is a square, so if width == height, it's a stop sign.

Thank you for the reply! I appreciate it

We have completed everything apart from the traffic signal detection (will finish that soon) thanks to your wonderful repository. Thank you so much, @hamuchiwa !

Have a look - https://drive.google.com/open?id=10Z7oCw4XpBe3b9kgd6YishomECpyB6iY

Yoganandhan01, 2020-03-20:
Hi Salman. I have also followed those steps but my car only goes on the straight that does not follow the lane. How can i solve the problem..?

Yoganandhan01, 2020-03-20:
@9salman7 can you help me.?

9salman7, 2020-03-21:
> Hi Salman. I have also followed those steps but my car only goes on the straight that does not follow the lane. How can i solve the problem..?

Hello. Are you following the OpenCV neural network model? I did that too and my car was also moving along just one direction. 
I'd suggest you to use a Keras neural network model. It works perfectly for me.

Yoganandhan01, 2020-03-22:
How can I implement Keras neural network into it.? Can you tell me some
examples.?
Yoganandhan A

9salman7, 2020-03-22:
You can refer to this Jupyter notebook:
https://github.com/9salman7/self-driving-car2/blob/master/kerasInfernus.ipynb

After training, you can load the model into your program to predict the directions

Yoganandhan01, 2020-04-01:
Thanks Salman..

Yoganandhan A

denisrexhepi, 2020-05-17:
Did the Keras model work?

Chinmay922, 2020-06-09:
I want to work on the same model but on MATLAB is it possible?

closed 2021-04-06 by hamuchiwa