hamuchiwa/AutoRCCar #249: rc_control_test.py problem !

Manou60xs, 2020-07-21:
Hi every one.
I had run rc_keyboard_control (arduino script) with windows 7
i didn't understand the connection between the raspberry and the arduino because when i run rc_control_test.py
in the raspbian it show me this message : 

pygame 1.9.4.post1
Hello from the pygame community. https://www.pygame.org/contribute.html
Traceback (most recent call last):
  File "/home/pi/Desktop/PFE/p/test/rc_control_test.py", line 73, in <module>
    RCTest()
  File "/home/pi/Desktop/PFE/p/test/rc_control_test.py", line 14, in __init__
    self.ser = serial.Serial("/dev/tty.usbmodem1421", 115200, timeout=1)    # mac
AttributeError: module 'serial' has no attribute 'Serial'

Please can anyone help me

Manou60xs, 2020-07-22:
@hamuchiwa  pls help , i need it for the license project

ShahirZain, 2020-07-22:
why you use RC for this.. . you can use bluetooth module

Manou60xs, 2020-07-27:
> Hi every one.
> I had run rc_keyboard_control (arduino script) with windows 7
> i didn't understand the connection between the raspberry and the arduino because when i run rc_control_test.py
> in the raspbian it show me this message :
> 
> pygame 1.9.4.post1
> Hello from the pygame community. https://www.pygame.org/contribute.html
> Traceback (most recent call last):
> File "/home/pi/Desktop/PFE/p/test/rc_control_test.py", line 73, in 
> RCTest()
> File "/home/pi/Desktop/PFE/p/test/rc_control_test.py", line 14, in **init**
> self.ser = serial.Serial("COM2", 115200, timeout=1) # mac
> AttributeError: module 'serial' has no attribute 'Serial'
> 
> Please can anyone help me

closed 2020-07-27 by Manou60xs