hamuchiwa/AutoRCCar #252: Help on sending arduino commnds to Raspberry Pi instead of host PC

theinfinitycorp, 2020-11-01:
Hey  guys,
This project is working flawless,however I am using a Arduino directly connected to the motor drivers instead of using a remote controller.
I'm able to run the car by connecting the  Arduino to pc and running the programs. 
I need help on doing the same by connecting the Arduino to the pi instead and send the commands  from the pc to the Pi ,so that the pi can control the arduino instead of needing to connect to the pc. 
Can you guys give pointers on how to achieve the same?
Thanks,

SaishaV, 2021-02-21:
Did you find a solution?

theinfinitycorp, 2021-02-25:
I did a workaround. I used my rpi as the host and used it to control the Arduino and run the programs.

closed 2021-04-06 by hamuchiwa